makerbase-mks/MKS-Power-Control #2: Printer doesn't boot when inserted fans into 12/24V connectors

artem9, 2020-06-13:
Today I installed the MKS PWC module into my 3D printer - TwoTrees Sapphire Pro. It has a mainboard MKS Robin Nano V1.2. I also made a fork from your "MKS Marlin 2.0 Firmware" repo, this is a configuration that I compiled and installed - https://github.com/artem9/Mks-Robin-Nano-Marlin2.0-Firmware/tree/sapphire-pro. Printer with this configuration can turn on/off (works as usual), but when I try to connect fans to the motherboard - it doesn't turn on anymore. 
This is the final build result - https://photos.app.goo.gl/SfETfBV3voTCGtVj6

mks-viva, 2020-10-22:
Is this problem solved?

mks-viva, 2020-10-27:
I will close this question temporarily.

closed 2020-10-27 by mks-viva