makerbase-mks/MKS-Power-Control #3: what does the MKS PWC 3.0 do in case of Power loss?

nictronik99, 2020-07-20:
What is the PWC capable of doing in the event of a power loss? Is it able to save the position if I enable Power loss recovery in Marlin that does not save the position on the SD layer by layer?

nictronik99, 2020-07-21:
Can anyone answer? The question is quite simple! @makerbase-mks

makerbase-mks, 2020-07-22:
Hi! MKS PWC does not affect the power loss function.

nictronik99, 2020-10-06:
But can't I use the pwc module with the ups module?

makerbase-mks, 2020-10-07:
Because MKS PWC and MKS UPS has detect function, for example, if AC power loss when printing, MKS PWC PW_DET has signal, AC power off and then MKS UPS detect AC power loss, Z axis move...

mks-viva, 2020-10-22:
@nictronik99 Is this problem solved?

nictronik99, 2020-10-22:
> @nictronik99 Is this problem solved?

Yes, thanks

closed 2020-10-22 by nictronik99