makerbase-mks/MKS-Power-Control #7: Question can I use MKS PWC and UPS together?

sima35, 2021-04-11:
I have a problem that both the PWC and the UPS are going to the same connector. I can't read your hub page
[image]

github051, 2021-04-12:
You cannot use both. Use only one

proterian, 2021-07-23:
why not?

javicarrera, 2021-07-27:
Yes, you can. 
My configuration: Robin Nano v1.2 - PWC v3 - MKS UPS 24V - LVGL_UI
I am using the LVGL UI, I haven't tried it with other interfaces.

Basically you wire the PWC as shown in the instructions, the UPS module is not connected to the main board (just 24V and AC).

And we make following changes:

configuration_adv.h:
Enable: POWER_LOSS_RECOVERY
Enable: BACKUP_POWER_SUPPLY
Uncomment: #define POWER_LOSS_ZRAISE       20 // in mm

Marlin\src\pins\stm32f1\pins_MKS_ROBIN_NANO.h:
Change this part.
Basically we uncomment  #define MKSPWC, add a new definition UPS.
which controls if KILL_PIN is declared or not and declares POWER_LOSS_PIN instead in order to allow us
to use both modules simultanously.

//
// Misc. Functions
//
#if HAS_TFT_LVGL_UI
  // LVGL
  […]

  #define MKSPWC 
  #define UPS 

  #ifdef MKSPWC
    #define SUICIDE_PIN                     PB2   
    #define SUICIDE_PIN_INVERTING          false  
    #ifndef UPS 
      #define KILL_PIN                        PA2   
      #define KILL_PIN_STATE                  true  
    #endif 
    
  #endif

  #define MT_DET_1_PIN                      PA4   
  #define MT_DET_2_PIN                      PE6   
  #define MT_DET_PIN_INVERTING             false  

  #define WIFI_IO0_PIN                      PC13  
  #define WIFI_IO1_PIN                      PC7   
  #define WIFI_RESET_PIN                  PA5   
#else 
  #define POWER_LOSS_PIN               PA2   
  #define PS_ON_PIN                          PB2   
  #define FIL_RUNOUT_PIN                PA4
  #define FIL_RUNOUT2_PIN               PE6
#endif 
#ifdef UPS
  #define POWER_LOSS_PIN                PA2   
#endif

Neispraven, 2021-12-05:
hi friend I have the same problem connecting PWC and UPS together my board is MKS SGEN LV 2.0 if it's not difficult for you help me figure it out in more detail ... here is my link to facebook https://www.facebook.com/profile.php? id = 100042489549041