makerbase-mks/MKS-Power-Control #10: MKS PWC v3 and MKS Robin E3 v1.1

nikolav75, 2021-08-19:
Hi
Can anyone help with the configuration and the wiring in this combination?
I'm using it with the main Marlin 2.0.9.1 and Ender 3.
I followed all instructions available online, but this thing just won't shut my printer down, no matter if I insert M81 at the end of the gcode or send it via Octoprint.
I'm following these instructions: https://github.com/makerbase-mks/MKS-Robin-E3-E3D/wiki/MKS_PWC
The 3 pins from the PWC's PW-DET are connected to the Robin's PB10. Then how should I connect the 3 pins from the PWC PW-OFF to the Robin's PA14, which has 2 pins only?
At the Robin's side I connected the green cable from PW-OFF to PA14 and the black one to ground. Where does the red one go?

This is what I get: https://youtu.be/vlqpeQNn3MI

nikolav75, 2021-08-19:
Sorted. I've swapped the PW_DET and PW_OFF wires at the PWC end. This issue is either safe to delete or leave as a monument of my stupidy and potentially to serve others with a similar issue.
Anyway, I think the documentation should better explain the cabling. It should state that the red cable from PW_OFF doesn't connect to anything.

mks-viva, 2021-08-19:
Thank you! As you said, I need to add a description, the role of the red cable.

mks-viva, 2021-08-19:
I has update wiki about how to use MKS PWC
[image]

nikolav75, 2021-08-19:
Actually, the PW_DET's red cable is connected (to 5V, I guess), as it comes with a single 3-pin connector and can't be split. It works anyway.

mks-viva, 2021-08-19:
MKS PWC does not actually need 5V, I will complete the prompt